mrcodetastic/ESP32-HUB75-MatrixPanel-DMA #845: Low horizontal Ghosting on half display

FishyFix, 2025-09-27:
I've been debugging my LED matrix project for quite some time, but I'm stuck on a persistent ghosting issue.

The matrix generally displays the intended content (yay!), but there's noticeable ghosting—especially visible when displaying a single pixel (see attached photo). The ghosting appears at a much lower brightness than the active pixel and occurs only in the same column, and only on one half of the display (the half where the pixel is located). I haven’t found anyone online describing this exact behavior.

Any ideas on what else I could try? I'd really appreciate any insights or suggestions!

[image]
[image]

 **Things I've tried:**

- Swapping in a different matrix of the same type
- 
- Increasing latch blanking
- 
- Inverting OE signal
- 
- decreasing clk speed
- 
- turning off clk phase
- 
- Rewiring ground connections

 

**Project Details:**

Board: ESP32-S2

LED Matrix: Adafruit 32x16 RGB LED Matrix

Wiring: Short (~5 cm) soldered wires connected to pin header for flatband cable

Library: ESP32-HUB75-MatrixPanel-I2S-DMA

**Test Code:**

```
#include "ESP32-HUB75-MatrixPanel-I2S-DMA.h"
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#define R1_PIN 4
#define G1_PIN 5
#define B1_PIN 18
#define R2_PIN 19
#define G2_PIN 21
#define B2_PIN 22
#define A_PIN 23
#define B_PIN 33
#define C_PIN 15
#define D_PIN -1
#define E_PIN -1 
#define LAT_PIN 25
#define OE_PIN 13
#define CLK_PIN 26  

#define PANEL_RES_X 32      // Number of pixels wide of each INDIVIDUAL panel module. 
#define PANEL_RES_Y 16     // Number of pixels tall of each INDIVIDUAL panel module.
#define PANEL_CHAIN 1      // Total number of panels chained one to another

MatrixPanel_I2S_DMA *dma_display = nullptr;

uint16_t myWHITE, myRED, myGREEN, myBLUE;

static const char *TAG = "RGB_TEST";

extern "C" void app_main() {
  HUB75_I2S_CFG::i2s_pins _pins={R1_PIN, G1_PIN, B1_PIN, R2_PIN, G2_PIN, B2_PIN, A_PIN, B_PIN, C_PIN, D_PIN, E_PIN, LAT_PIN, OE_PIN, CLK_PIN};
  HUB75_I2S_CFG mxconfig(
    PANEL_RES_X,   // module width
    PANEL_RES_Y,   // module height
    PANEL_CHAIN,    // Chain length
    _pins // pin mapping
  );

  // mxconfig.latch_blanking = 4;
  // mxconfig.i2sspeed = HUB75_I2S_CFG::HZ_20M;
  // mxconfig.driver = HUB75_I2S_CFG::SHIFTREG;
  // mxconfig.clkphase = false;
  // mxconfig.double_buff = true;
  // mxconfig.min_refresh_rate = 120;

  dma_display = new MatrixPanel_I2S_DMA(mxconfig);
  dma_display->begin();
  dma_display->setBrightness8(50); // 50 is currently max
  dma_display->clearScreen();
  
  myWHITE = dma_display->color565(255, 255, 255);
  
  dma_display->clearScreen();
  
  dma_display->drawPixel(16, 8, myWHITE);

  dma_display->drawPixel(7, 5, myWHITE);
}
```

FishyFix, 2025-10-04:
Another Example with some colors.

[image]

mrcodetastic, 2025-10-25:
You might need a logic level converter between the ESP32 and the HUB75 as some panels don't play well when driven by 3.3V (they're supposed to be driven by a 5V signal).

I know this is a pain, but it would be interesting to know if you get ghosting with the PixelMatrix/SmartMatrix library as well.

FishyFix, 2025-10-25:
Hi thanks for your help. I am actually driving with a 5V(4A) power supply. Now that you mentioed the PixelMatrix/SmartMatrix library, I had used an Arduino before with the Adafruit library and I don't remember seeing this issue. I will try the PixelMatrix/SmartMatrix library when I find some time and get back.

mrcodetastic, 2026-04-21:
Did you end up finding a fix?